charely6/mbotmake #2: Mbotmake does not set heat temperature

kk-kk-k, 2025-03-08:
**Describe the bug**
Your nice program does not set nozzle heat temperature, leading to cool down while printing and filament stuck on the nozzle

**To Reproduce**
Steps to reproduce the behavior:
1. Slice your project normally with temperature around 200 C
2. Run mbotmake
3. Print the .makerbot file
4. See the nozzle cooling down

**Expected behavior**
I expect the nozzle to have temperature around 200 C, not to cool down and print

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Arch linux
 - Version latest

**Additional context**
In your script, I would except not to set temperature to 0 in line 16

Good luck

antonysigma, 2025-04-11:
In my case (Prusa slicer), it is caused by the manual ending gcode (i.e. shutting off extruder header with temperature value 0) overriding the previous temperature values. Try out this ad hoc fix: https://github.com/charely6/mbotmake/compare/master...antonysigma:mbotmake:master

antonysigma, 2025-04-24:
Resolved in the experimental branch since #10 . Closing.

closed 2025-04-24 by antonysigma