charely6/mbotmake #5: v2 Development roadmap

antonysigma, 2025-04-11:
@charely6 Help wanted to polish the v2 `mbotmake2` in the git branch `experimental`:

Roadmap:

- [X] Set up Github actions to verify the code.
- [X] Generate the universal Python wheel files.
- [X] (For windows users only) Invoke Nuitka to build a standalone exe app.
- [ ] Add more test cases to the tests/ folder.

Examples of the additional test cases:

- [ ] The fan toggle state. Ensure the fan is turned on with an explicit command, ahead of setting Fan PWM value. Write a `pytest` logic in the `tests/` folder.
- [ ] Verify that the path `mbotmake2/templates/*.json` exists, after running `uv pip install -U .`.
- [ ] Cura gcode test cases. Write a `pytest` script in the `tests/` folder that loads the `testcases/*cura*/cube.gcode` input and process it without error.
- [ ] Uncomment the assertions `xrel` and `yrel` to be less than user defined constraints.

dzatoah, 2025-04-25:
Awesome, I'm happy that this project is still maintained and new features are in development. I'll definitively test this, when I have time.

I recently acquired a free Makerbot Replicator 5th Gen with Smart Extruder +.

I quickly want to mention that there is an unofficial web converter available too: https://makerbot.online/slicer/converter/
I'm not affiliated in any shape or form.

Which slicer do you currently use for development? I use Cura. Is it possible to add this script as a post processing script or something? Is there maybe a pre-written config file?

Thanks in advance and especially for your already done work 🙏🙏

antonysigma, 2025-04-25:
I am glad you like it. Please do contribute test cases by uploading/submitting example gcode files from your slicer. Refer to #14 for details.

> I quickly want to mention that there is an unofficial web converter available too:  https://makerbot.online/slicer/converter/
> I'm not affiliated in any shape or form.

Personally, I am afraid that `mbotmake`, just like other *right to repair* movement, is bounded by the [Goldilocks principle](https://en.wikipedia.org/wiki/Goldilocks_principle). Not too buggy, and not too famous. It must carry along just the right amount of software bugs for `mbotmake` to be sustainable.

If the software is too buggy, nobody will use it or improve it. If too famous, then the manufacturers could be losing profits, motivating them to make breaking changes to the printer drivers.