charely6/mbotmake #15: mbotmake2.exe crashes immediately with error.

dzatoah, 2025-04-25:
I've downloaded the latest mbotmake2.exe artifact from the last pipeline/action run, but now I've encountered an issue:

`Error, couldn't launch child ([Error 225] (null))`

I couldn't find anything on the internet regarding this error. Very strange. I'm using Windows 10. `10.0.19045 Build 19045`.
Administrator rights are not helping.

antonysigma, 2025-04-25:
Possible duplicate of #9 .

dzatoah, 2025-04-25:
Yeah, but I think the changes should already be included in the exe, I used this artifact: https://github.com/charely6/mbotmake/actions/runs/14633427806/job/41059813720

antonysigma, 2025-04-29:
Deferring to @charely6  for Windows specific issues. To file a bug report, please download the xyz calibration STL part from Thingiverse for testing, and then follow the instructions at #14 to produce a minimal reproducible error (MRE) for us. https://www.thingiverse.com/thing:1278865

antonysigma, 2025-05-06:
Duplicate of #16 . Closing.

closed 2025-05-06 by antonysigma