charely6/mbotmake #23: Why only absolute move command? Is relative move supported?

antonysigma, 2025-05-20:
I've tested with Makerbot Replicator 5th gen.,with Smart Extruder+, with Version 2.6.3 and I confirm, that the .makerbot I generated shows the biggest thumbnail too.

Maybe you already saw that the gcode contains non supported code. I had to patch the software a little to compensate for that....

```patch
From fa8c44a992b3f54728c33a454ff8caceb324e74c Mon Sep 17 00:00:00 2001
From: Daniel Teichmann <Daniel Teichmann>
Date: Fri, 9 May 2025 21:20:44 +0200
Subject: [PATCH] patch to bypass gcode restrictions

---
 
         # Checking toolpath
-        assert transformer.is_extrusion_absolute, "Fatal: Relative extrusion position detected."
-        assert transformer.is_xyz_absolute, "Fatal: Relative xyz position detected."
+        # TODO: Why this is this asserted? Is it not supported?
+        #assert transformer.is_extrusion_absolute, "Fatal: Relative extrusion position detected."
+        #assert transformer.is_xyz_absolute,       "Fatal: Relative xyz position detected."
```

_Originally posted by @dzatoah in https://github.com/charely6/mbotmake/issues/21#issuecomment-2867679388_

antonysigma, 2025-05-20:
@charely6 can inform you the original design intent for limiting `mbotmake` to only support absolute move.

My personal take on the question is that position encoded JSON format, required by Makerbot firmware, suffers from round off errors due to the JSON requirement of floating point representation. We don't want our Python code to accumulate round off errors. It is a job better handled by the slicer not python nor json.

charely6, 2025-05-20:
Honestly I didn't write it originally. I just saw that was an issue when I tried to use it originally so I noted it. If we can use relative moves in the makerbot files I would totally say we should do that but otherwise I think having the slicer do it as absolute movements is fine.